ukaea/paramak #32: Images for the documentation and Readme

Shimwell, 2020-07-01:
The images which are currently hosted on google drive and are not loading up every time the Readme is loaded.

Remi has shown us another way using github issues to host images which we can try :-)

[image]
[image]
[image]
[image]

Shimwell, 2020-07-01:
[image]

billingsley-john, 2020-07-01:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Shimwell, 2020-07-01:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

RemiTheWarrior, 2020-07-01:
Hi all, you don't even have to create the whole issue.
Just open a new one, copy paste your image, copy the URL in your readme, cancel the issue.

Shimwell, 2020-07-01:
[image]
[image]

Shimwell, 2020-07-01:
[image]

Shimwell, 2020-07-02:
[image]

Shimwell, 2020-07-02:
[image]

Shimwell, 2020-07-02:
[image]
CenterColumnShieldCircular

Shimwell, 2020-07-02:
[image]

Shimwell, 2020-07-04:
[image]

blanket constant thickness horizontal ends

Shimwell, 2020-07-07:
[image]

billingsley-john, 2020-07-08:
[image]
[image]
[image]
[image]

billingsley-john, 2020-07-08:
[image]

billingsley-john, 2020-07-08:
[image]

billingsley-john, 2020-07-09:
[image]

billingsley-john, 2020-07-09:
Adding new images of parametric_shapes which are all approximately the same size. This makes it easier to create gifs and size tables correctly

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

billingsley-john, 2020-07-09:
[image]
[image]
[image]

billingsley-john, 2020-07-09:
[image]

Shimwell, 2020-07-09:
[image]

Shimwell, 2020-07-12:
[image]

Shimwell, 2020-07-13:
[image]
new plasma

billingsley-john, 2020-07-21:
Updated sweep shapes 

[image]
[image]
[image]
[image]

billingsley-john, 2020-07-21:
[image]

better extrude sweep shape

billingsley-john, 2020-07-21:
[image]

RemiTheWarrior, 2020-07-22:
[image]

RemiTheWarrior, 2020-07-22:
[image]

billingsley-john, 2020-07-23:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

billingsley-john, 2020-07-29:
[image]

billingsley-john, 2020-07-29:
[image]

Shimwell, 2020-07-30:
[image]
[image]

billingsley-john, 2020-07-30:
[image]

billingsley-john, 2020-07-30:
[image]

Shimwell, 2020-08-03:
[image]

billingsley-john, 2020-08-03:
[image]

billingsley-john, 2020-08-03:
[image]

Shimwell, 2020-08-05:
[image]

Shimwell, 2020-08-05:
[image]

Shimwell, 2020-08-05:
[image]

billingsley-john, 2020-08-26:
[image]

billingsley-john, 2020-09-02:
[image]

billingsley-john, 2020-09-03:
[image]
triple arc

billingsley-john, 2020-09-03:
[image]
princeton plasma d

Shimwell, 2020-09-11:
[image]

Shimwell, 2020-09-11:
[image]

Shimwell, 2020-09-18:
[image]

Shimwell, 2020-09-18:
[image]

Shimwell, 2020-09-18:
[image]

Shimwell, 2020-09-21:
[image]

Shimwell, 2020-09-21:
[image]

Shimwell, 2020-09-22:
[image]

Shimwell, 2020-09-24:
[image]

Shimwell, 2020-09-24:
[image]

Shimwell, 2020-09-25:
[image]

[image]

Shimwell, 2020-09-29:
[image]

Shimwell, 2020-09-29:
[image]

Shimwell, 2020-09-29:
[image]

Shimwell, 2020-09-29:
[image]
[image]

Shimwell, 2020-09-30:
[image]
[image]

Shimwell, 2020-09-30:
[image]

[18 more replies]

closed 2020-10-26 by Shimwell