ukaea/paramak #59: Enhancement: analytical expression for plasma object

RemiTheWarrior, 2020-07-12:
If I understood correctly, the component plasma is created from 2 arcs.

Why not using the actual analytical expression for the plasma boundary ? (based on elongation, triangularity, minor and major radii)

[image]

This would give a smoother and perhaps more accurate shape ?

Shimwell, 2020-07-12:
That looks like a nicer method. I guess that makes supporting double and single null plasmas easier as well.

The plasma is one of the oldest shapes so it is probably can be done in a better way.

It would be handy for other shapes that try to keep a constant offset if the X point (X and Z) coordinates are also available from the plasma objects

RemiTheWarrior, 2020-07-12:
The trick is to find the correct equations. See [this paper](https://fire.pppl.gov/ARIES_AT_Equilibrium_kessel.pdf)

Shimwell, 2020-07-12:
Equation 7 looks good

RemiTheWarrior, 2020-07-12:
Is it "double null friendly" ?

RemiTheWarrior, 2020-07-13:
Done in #64

closed 2020-07-13 by RemiTheWarrior