ukaea/paramak #60: improved images that show parameters

Shimwell, 2020-07-12:
The images for parametric components, shapes and reactors are nice but they don't show how the parameters change the shape of the models

One option is to make svg images using the exportsvg method then convert this to a png then add annotations to this image using  latex

I had a quick go at this to see how it looks, perhaps this or something like this could be something we consider adding to the documentation in the future.

[image]

[image]

billingsley-john, 2020-07-13:
I think this would be a good addition, especially for the documentation

RemiTheWarrior, 2020-07-13:
@Shimwell good idea, perhaps add the origin location.

Shimwell, 2020-07-13:
[image]

Added a Z axis and rotation angle

Shimwell, 2020-07-13:
I have been using https://www.mathcha.io/ to make Latex images and it speeds up the process quite a lot

Shimwell, 2020-08-06:
The documentation now has lots of parametric components and quite a lot have nice diagrams

- [x] BlanketConstantThicknessArcH
- [x] BlanketConstantThicknessArcV
- [ ] BlanketFP
- [x] CenterColumnShieldCylinder
- [x] CenterColumnShieldHyperbola
- [x] CenterColumnShieldFlatTopHyperbola
- [x] CenterColumnShieldFlatTopCircular
- [ ] CenterColumnShieldPlasmaHyperbola
- [ ] ITERtypeDivertor
- [ ] DivertorITERNoDome
- [x] InnerTfCoilsCircular
- [x] InnerTfCoilsFlat
- [ ] PoloidalFieldCoil
- [ ] PoloidalFieldCoilCaseFC
- [ ] PoloidalFieldCoilCase
- [ ] Plasma
- [ ] PlasmaFromPoints
- [ ] ToroidalFieldCoilRectangle
- [ ] ToroidalFieldCoilCoatHanger

Shimwell, 2020-09-25:
[image]

Shimwell, 2020-10-01:
I've added a few more yesterday, but we have more components now.

Easy ones to do are the PoloidalFieldCoil and the Plasma

And I somehow keen forgetting to add the origin even with the reminders, sorry about that :-)

Shimwell, 2020-10-01:
Plasma and Plasma from points done

Shimwell, 2020-10-01:
ToroidalFieldCoilTripleArc (thanks @RemiTheWarrior), PoloidalFieldCoilCase and PoloidalFieldCoil all done

Shimwell, 2020-10-26:
Closing this issue for the same reasons as #32 . Please add images to the PR instead

closed 2020-10-26 by Shimwell