ukaea/paramak #61: BUG: plasma is not displayed if rotation_angle is not specified

RemiTheWarrior, 2020-07-12:
Code to reproduce :
```python
import paramak

plasma = paramak.Plasma(minor_radius=200, major_radius=620, triangularity=0.55, elongation=1.85)
plasma.solid
```

When the argument `rotation_angle` is not given, the `.solid` method doesn't display the plasma.

Shimwell, 2020-07-12:
Ah yes this is a strange one.

When the rotation angle is not specified then it defaults to 360 degrees. This makes a complete spline where the start and end surface connect. This sometimes breaks but a rotation angle of less than 360 (e.g. 359) works in the visualization

There is a bug in OCC 7.3 which cascades to CadQuery and onto the Paramak when such shapes don't get created.

However it works if you export the stp file then everything appears to work

anyway the bug is getting fixed upstream and more details can be found here
https://github.com/CadQuery/cadquery/issues/259

It might take a while for the fix to propagate through the various stages and appear in the conda install

sorry about this

RemiTheWarrior, 2020-08-27:
@Shimwell the update of the docker image seems to have fixed this. If the new image is used in the CI then I can close this.

closed 2020-08-27 by RemiTheWarrior