ukaea/paramak #84: Missing points impact html and png export

Shimwell, 2020-07-16:
when exporting html scatter graphs of the reactor or exporting 2d_images the methods rely on points in the object.

These points are not fully representative of the objects just to the interconnections sometimes being circles or splines

One example is when creating a blanket from an arc then the three points show up as a triangle 
[image]

One solution for this is to get matplotlib to use the connections between points, another solution is to faceted the shapes using the method described here

https://groups.google.com/forum/#!searchin/cadquery/spline%7Csort:date/cadquery/Ia84dtcnCo0/CCvVIl1bBQAJ

I think @billingsley-john might be interested as this also causes problems for the sweep circle operations

We could also move to a completely different way of viewing the objects using a slice through the CAD model or make use of openmc plotting

I'm keen on the faceting idea as I think it could come in useful when trying to export splineless CAD

Shimwell, 2020-07-16:
Also the matplotlib method really doesn't work when the workplane is changed, the left hand side triangle in the image above is the inner tf coil

billingsley-john, 2020-07-20:
Faceting the shapes seems like the better option with the added benefits for exporting splineless CAD, however, I appreciate it is much more complex

Shimwell, 2021-01-25:
solved with #693

closed 2021-01-25 by Shimwell