ukaea/paramak #85: Contributions: templates for PR and bug report

RemiTheWarrior, 2020-07-17:
A low effort addition that will facilitate contribution (imho)

Shimwell, 2020-07-17:
Any preference to the template used for PR

Here are a few to choose from 

https://github.com/devspace/awesome-github-templates#rocket-templates-for-pull-requests

RemiTheWarrior, 2020-07-17:
https://github.com/appium/appium/blob/master/.github/PULL_REQUEST_TEMPLATE.md

I like this one. I think the check list is the most important thing

Shimwell, 2020-07-17:
We might have to remove a few check boxes as we don't have a contribution guide or a cli.

I can see how it would help keep the code tidy.

Only downside is that it slows down commits and might deter contributions. But people can always ignore the template if they are in a rush.

I can put in a PR for the template using the template just to get some practice

RemiTheWarrior, 2020-07-17:
@Shimwell sure some aren't usefull. Well as it's not mandatory, it won't be limiting. But yeah it helps (even for internal developpers) to keep in mind the things to do (like docs) before merging

RemiTheWarrior, 2020-07-17:
@Shimwell current vs new behaviour can also be a nice section. It's up to you

Shimwell, 2020-09-03:
PR template is now in develop. It will work once it gets to the default branch (main). @billingsley-john is merging into main today so it should be up and running soon.

Thanks for the suggestion

closed 2020-09-10 by RemiTheWarrior