ukaea/paramak #94: Improved arguments for shape.py

Shimwell, 2020-07-19:
Currently we have a few keyword arguments for shape.py

Adding new keywords such as the recently added tet_mesh and physical_groups to the keywords propagates to the super statements which means updating all the classes that inherit. 

Therefore they have been added as attributes but not as arguments.

This is a shame as uses would have to may shapes then set attributes after the shape object is initialized. Users can't currently pass extra keyword on creation of the shape object which is a shame.

This can be fixed to allow users to add attributes or keyword args to the object on creation
https://treyhunner.com/2018/04/keyword-arguments-in-python/

billingsley-john, 2020-08-04:
I will take a look at this

billingsley-john, 2020-08-04:
Notes:
kwargs need to be added at the user-level classes

billingsley-john, 2020-08-11:
tet_mesh and physical_groups have been added to the default_dict of user-level classes, and can be specified as arguments upon construction

closed 2020-08-11 by billingsley-john