ukaea/paramak #113: Conda install

Shimwell, 2020-07-23:
The current install process is a bit messy with a mix of conda and pip

Lets make a conda install and make it easier for people to get started

There are a few how to guides e.g.
https://medium.com/@giswqs/building-a-conda-package-and-uploading-it-to-anaconda-cloud-6a3abd1c5c52

Shimwell, 2020-08-03:
Perhaps we should go for **conda** **forge** instead of anaconda cloud as both DAGMC and OpenMC use conda forge and these are codes we aim to connect to

Shimwell, 2020-08-03:
A PR has been created to keep track of this effort https://github.com/ukaea/paramak/pull/143

Shimwell, 2020-11-22:
Looks like #143 has got a bit out of date.

The import files from that PR are: 
build.sh
bld.bat
environment.yml 
meta.yaml
update_conda.sh 

We I get some time I shall close this PR and see if I can make the conda install

Shimwell, 2021-02-01:
I am making progress with this issue.
Conda appears to build a package for all os systems
this is the branch I'm working on
https://github.com/ukaea/paramak/tree/conda_install

Shimwell, 2021-02-02:
Package creation and upload appears to be working but the install is not quite working

[image]

jwscook, 2021-02-05:
What's the output with `-v` for verbosity in `conda install -v -c openmcworkshop paramak`?

Shimwell, 2021-02-05:
Thanks for the suggestion

[image]

jwscook, 2021-02-05:
Looks like paramak doesn't allow python version 3.7 or 3.8 for that matter. Can you try it with a 3.6 miniconda container?

Shimwell, 2021-02-05:
Something for next week but just wanted to post these finds before I forget

```
sudo docker run -it continuumio/miniconda3
conda create --name myenv python=3.6
conda activate myenv
conda install -v -c openmcworkshop paramak
```

[image]

peter-t-fox, 2021-02-09:
I've tried to resolve this today, but have made essentially no progress. Things I've tried:

* Installing paramak in a clean Python 3.6, 3.7 and 3.8 environment. In each case, the install fails with the same error message, indicating that paramak is not compatible with the version of python under which it is being installed.
* When running ``conda install``, the correct version of paramak seems to be selected in each case, but you can be explicit by doing e.g.``conda install -c openmcworkshop paramak=0.2.3=py36hfeaa757_0``.
* The metadata for each version of paramak seems to be set up correctly. For example, running ``conda search --info -c openmcworkshop paramak=0.2.3=py36hfeaa757_0`` gives:

```
file name   : paramak-0.2.3-py36hfeaa757_0.tar.bz2
name        : paramak
version     : 0.2.3
build       : py36hfeaa757_0
build number: 0
size        : 190.6 MB
license     : MIT
subdir      : linux-64
url         : https://conda.anaconda.org/openmcworkshop/linux-64/paramak-0.2.3-py36hfeaa757_0.tar.bz2
md5         : 9ba5566e753d15ff63c056108533bf30
timestamp   : 2021-02-05 13:55:27 UTC
dependencies: 
  - cadquery
  - matplotlib
  - mpmath
  - neutronics_material_maker
  - numpy
  - plasmaboundaries
  - plotly
  - pytest-cov
  - python >=3.6,<3.7.0a0
  - python_abi 3.6.* *_cp36m
  - scipy
  - sympy
```
which shows the expected python range. This suggests that it might be a conflict in the dependencies rather than in the core paramak package itself.

* Each of the dependencies can be installed individually without difficulty, **except** ``cadquery``, which fails under Python 3.8, interestingly with a similar error message.
* By building the ``paramak`` package and putting it in a custom conda channel on my local filesystem, I *can* get it to install correctly under Python 3.6 (haven't tried with the other versions, but likely it would work there too). It is not obvious why there would be a difference in these two approaches, as the installation tarball should be identical.

In general, it seems to be very difficult to extract information from conda concerning how it determines when there is a specification mismatch, and what packages in the dependency list contribute to any element of the specification. Without this information, it's very difficult to see what's going on.

Shimwell, 2021-02-09:
Sounds like a good investigation, thanks for taking a look at this. The puzzle continues. It will be all the more satisfying when solved.

I wonder if it is getting the old version of cadquery before the project shifted away from freecad

Shimwell, 2021-02-09:
Ignore that last thought of mine, looks like the channel priority is correct. channel cadquery is higher priority than channel conda-forge in the conda_build.sh

Shimwell, 2021-02-11:
I've been trying this again but this time in a new branch with the latest features from the parmak as the old branch was getting a bit out of date and have a few conflicts.

https://github.com/ukaea/paramak/tree/conda_install_2