ukaea/paramak #119: tidy up the readme

Shimwell, 2020-07-28:
the home page has a few out of date things

Usage - Shape creation should include images and be renamed to something like Usage - parametric shape

Usage - Creating a Plasma should be renamed to something like this Usage - parametric shape creation

Import statement is wrong for the plasma

Usage - Reactor creation should include images and be changed to Usage - parametric reactor

the parametric reactor example uses add which no longer exists

utils.py should also be added to the "Core modules" section

Perhaps the long list of python classes could be removed from the index page

billingsley-john, 2020-07-29:
Update - TODO:
Add parametric reactors section
utils.py should also be added to the "Core modules" section
Perhaps the long list of python classes could be removed from the index page
Add images to examples

billingsley-john, 2020-08-04:
Update - TODO:
utils.py should also be added to the "Core modules" section

closed 2020-08-04 by billingsley-john