ukaea/paramak #142: A blanket which has a start and stop thickness

Shimwell, 2020-08-02:
All of the blankets are Constant Thicknesses which is great but occasionally there is a need for a variable thickness blanket

For example the EU DEMO blanket has a thicker outboard blanket than the inboard blanket https://images.app.goo.gl/5FCPGWVKosSfgJ7Z7

perhaps thickness could allow a list of two thickness and then linear interpolation between the 2 points or a start_thickness and stop_thickness could be used instead

Shimwell, 2020-08-03:
Perhaps a variable offset from the plasma as well

RemiTheWarrior, 2020-08-03:
@Shimwell How about a component with thickness distribution along the poloidal section ?

Shimwell, 2020-08-03:
Yes that sounds even better

RemiTheWarrior, 2020-08-03:
@Shimwell well in that case this can be achieved by adapting the current blanket and allow the thickness argument to be a callable (a function of poloidal angle).

Shimwell, 2020-08-03:
this would be super flexible, perhaps a bit harder for the user compared to just two numbers [start_thicknes, stop_thickness]

RemiTheWarrior, 2020-08-04:
as often: why not do both ?
The thickness parameter can be a float (constant thickness), a tuple of 2 floats (thickness varying linearly between start_thickness and stop_thickness), or a callable (function of angle).

Shimwell, 2020-08-04:
That would be the ultimate solution.

RemiTheWarrior, 2020-08-04:
This would be needed if a user would want to make this kind of blanket
[image]

RemiTheWarrior, 2020-08-06:
Done in #158 this can be closed

closed 2020-08-25 by Shimwell