ukaea/paramak #148: BUG: export_3d_image method not working

RemiTheWarrior, 2020-08-04:
In #138 we added some more tests in order to improve the code coverage but some are failing. In particular the one involving `export_3d_image` method.

See [this build](https://app.circleci.com/pipelines/github/ukaea/paramak/345/workflows/92ca15a7-67fe-4d0a-ae87-b12198ae0c01/jobs/352)

Shimwell, 2020-08-04:
What do people think about this. Shall we remove this export 3d render option and perhaps convert it into and example. It doesn't appear to work with docker image either. Now that we have SVG export working perhaps this is not needed as much as before. I don't mind either way.

billingsley-john, 2020-08-04:
I wouldn't mind removing the 3d render, given the additional visualization methods that have been added

Shimwell, 2020-08-06:
I'm tempted to transfer the rendering code into an example and then merge the PR #150 
So the export_stl method is now implemented so this helps with exporting the geometry for rendering

Shimwell, 2020-08-25:
I think this has now been fixed in #150 #204 and #203

PR #150 (2020-08-24): export_3d_image removed
This PR resolves #148 

Coverage : 95%.
closed 2020-08-26 by RemiTheWarrior