ukaea/paramak #164: Single null designs

Shimwell, 2020-08-06:
Would be great to have a parametric reactor that just has a single divertor

This could probably just be a small change to the existing BallReactor and SubmersionTokamak that takes an ['upper'],['lower'] or ['upper', 'lower'] argument and then makes the diverter envelop a little shorter. The divertor envelop is used in conjunction with the intersect feature and th firstwall geometry to make the divertor. The height of the divertor envelope can be reduced so that it only covers the upper or lower part of the firstwall

billingsley-john, 2020-08-17:
To create a single null design, blanket_fw_rear_wall_envelope must be changed to include only the top or bottom half of the envelope. This cannot be done via BlanketConstantThicknessArcV because the correct spline curvature must be preserved, meaning the 'full' envelope must be constructed. An easy solution would be to simply cut z > 0 or z < 0 with a large box, however, this involves using native CadQuery commands.

A better solution could be to add something like an optional start/stop angle to BlanketConstantThicknessArcV so that segments of this shape can be created. This would allow you to create a shape with the correct spline curvature but extract the upper or lower half to create the envelope for divertor creation. However, I realize that this may take the class further away from its initial purpose as you can lose the vertical 'ends'. (same could be applied to BlanketConstantThicknessArcH)

Shimwell, 2020-08-17:
Lines 431-439 in https://github.com/ukaea/paramak/blob/develop/paramak/parametric_reactors/ball_reactor.py create the divertor. Perhaps change these so that cylinder made only overlaps lower blanket instead of both upper and lower as it does now

billingsley-john, 2020-08-19:
Update:
Work has commenced on splitting up the create_components() method in ball_reactor.py so that components are created separately in different methods. This currently requires a lot of returning and passing of objects/attributes to different methods.
At the moment, only models including the optional pf and tf parameters can be built, when they are removed, errors occur as some functions expect the values to be passed.
I think using *args and **kwargs could sort this because any required parameters could be passed, but the function would not error if it didn't get passed the optional pf and tf arguments.

billingsley-john, 2020-08-19:
Separating create_components() into different methods allows us to rewrite the make_divertor() method for a single_null design, without having to rewrite the whole create_components() method, i.e. we can just inherit the methods for all other component creation.

billingsley-john, 2020-08-19:
Update:

Branch = paramak_single_null_new contains a rewrite of ball_reactor() which separates make_components() into a set of separate methods. The passing of arguments is messy and should be refined, but this cannot be done efficiently with *args or **kwargs.
The branch also contains single_null_new.py which is a new parametric single null reactor. This inherits from BallReactor but defines its own make_components() and make_divertor() methods which are used to create the single null reactor.

These scripts are very messy at the moment and will need some refinement before merging, however, they do show a way of rewriting ball_reactor so that we can reuse its methods and create different reactors more easily, without having to rewrite the entire make_components() method, as mentioned earlier.

billingsley-john, 2020-08-20:
Resolved in #199

closed 2020-08-26 by billingsley-john