ukaea/paramak #174: Adding more plasma plots

Shimwell, 2020-08-06:
The [make_plasmas.py](https://github.com/ukaea/paramak/blob/develop/examples/example_parametric_components/make_plasmas.py#L134-L141) example script has 4 basic plasma and one more [advanced plasma.](
https://github.com/ukaea/paramak/blob/45443e5696c7f6519ba1972570fb809c33236163/examples/example_parametric_components/make_plasmas.py#L134-L141)

Could this script be upgraded to plot the basic plasmas on te left by multiplying their x values by -1 and the advance plasmas on the right

3 more advanced plasma also be added to the script using these input [parameters](https://github.com/RemiTheWarrior/plasma-boundaries/blob/master/plasmaboundaries/parameters.py)

@billingsley-john

RemiTheWarrior, 2020-08-07:
@Shimwell @billingsley-john I can help with that if you want. Care must be taken cause the parameters in plasmaboundaries are rationalised by major_radius.

billingsley-john, 2020-08-18:
#197 plots basic plasmas on the left and plasmaboundaries plasmas on the right. Major radius and minor radius are required before spheromak equilibrium plasma, NSTX_double_null plasma and NSTX_single_null plasma can be added.

Do you know these values for these plasmas @RemiTheWarrior?

RemiTheWarrior, 2020-08-19:
@billingsley-john I'm afraid I don't but will dig this up

RemiTheWarrior, 2020-08-19:
@billingsley-john according to [wikipedia](https://en.wikipedia.org/wiki/National_Spherical_Torus_Experiment), the major radius is 0.85m and minor radius is 0.68m.

billingsley-john, 2020-08-20:
Resolved in #197

closed 2020-08-20 by billingsley-john