ukaea/paramak #186: make_faceted_neutronics_model.py script error

billingsley-john, 2020-08-11:
find_number_of_volumes_in_each_step_file() function cannot be split over lines, must be one complete line

CURRENT:

geometry_details = find_number_of_volumes_in_each_step_file(
    geometry_details, str(os.path.abspath("."))
)

FIX TODO:

geometry_details = find_number_of_volumes_in_each_step_file(geometry_details, str(os.path.abspath(".")))

Shimwell, 2020-08-11:
Ah I'm sorry this is probably my fault running the pep8 processor over the whole code base. We can revert these two Trelis scripts

billingsley-john, 2020-08-12:
Script requires line continuation character ("\") to be explicitly specified for trelis

billingsley-john, 2020-08-12:
autopep8 removes newline characters, so this is not a viable solution

billingsley-john, 2020-08-13:
the argument --exclude="make_faceteted_neutronics_model.py" can be used locally to ignore the trelis script, however, this does not seem to work when added to the autopep8.yml file.

billingsley-john, 2020-08-13:
Resolved across #188, #189, #191, and #193.

yml file required --exclude=make_faceteted_neutronics_model.py (i.e. without " ")

PR #189 (2020-08-13): added trelis file to be ignored
PR to add make_faceteted_neutronics_model.py to the list of files to be ignored by autopep8. This resolves #186
PR #188 (2020-08-13): added newline character
This is an alternative solution to #187 for fixing the trelis make_faceteted_neutronics_model.py script. This does not require a rollback of the autopep8 formatting. Resolves #186
closed 2020-08-13 by billingsley-john