ukaea/paramak #205: Princeton-D TF coil

RemiTheWarrior, 2020-08-26:
Thanks to the system of differential equations described [here](https://repository.lib.ncsu.edu/bitstream/handle/1840.20/27004/N2.1-6.pdf?sequence=1), Princeton-D TF coils can be added to the Paramak.

RemiTheWarrior, 2020-08-26:
@Shimwell @billingsley-john what do you think the input parameters should be ?

RemiTheWarrior, 2020-08-26:
[image]
So far 2 input parameters must be chosen out of 3 which are R1 (orange line), R2 (green line) and R0 (red line).

Shimwell, 2020-08-26:
Super, 3 radial build parameters, I guess the red one can be chosen to match the plasma major radius
Does the thickness of the coil feature as a parameter

RemiTheWarrior, 2020-08-26:
Well we need to choose 2 of these 3 parameters and the third one will be computed based on the two others.

Yes the thickness of the coil is a parameter see #207

RemiTheWarrior, 2020-08-31:
Resolved in #207

closed 2020-08-31 by RemiTheWarrior