ukaea/paramak #216: BallReactor construction error

billingsley-john, 2020-09-01:
BallReactor has intermittent construction error, either defined as StandardConstructionError or a 'NoneType' error. Error can be reproduced both locally and within the openmcworkshop/paramak_dependencies docker image

Shimwell, 2020-09-01:
Thanks for trying the docker environment.
Wondering if this is related to the use of circles that was also causing trouble in the Trelis workflow. We could replace the use of blankets from arcs with the better blanket geometry that uses splines

billingsley-john, 2020-09-01:
This issue seems to be related to rotation_angle. Tests pass when models are constructed with rotation_angle < 360, but sometimes fail when rotation_angle = 360.
This looks like an internal CadQuery error.
Issue #212 may also be related to this

billingsley-john, 2020-09-01:
Error is also present for the SubmersionTokamak. This was not picked up earlier as all tests had rotation_angles < 360

Shimwell, 2020-09-01:
Is it possible to isolate this to a single component.

billingsley-john, 2020-09-01:
In the case of the BallReactor, it is always when constructing the firstwall, blanket or blanket rear casing. Traceback indicates the error occurs when components are fused or cut

RemiTheWarrior, 2020-09-01:
@billingsley-john I also found out that if the connections between points (in the case of #212 for the BlanketFP) is `"straight"` everywhere and without `"spline"` connections, then this doesn't appear.

This would indicate it's due to a mismatch caused by the fact that spline connections are not exact.

RemDelaporteMathurin, 2020-11-09:
is this related to #445 ?

closed 2020-11-23 by RemDelaporteMathurin