ukaea/paramak #235: improved readme

Shimwell, 2020-09-07:
We could do with a more focused Readme.

I'm assuming GitHub is where developers and source code people will go

My concern is that they see the examples and don't go to the documentation

So I'm thinking we should move the examples over and or make the examples in the documentation better.

RemiTheWarrior, 2020-09-07:
@Shimwell if my opinion is of any help, I'd think that the Github page is still a good showcase for the code for people who just google "Paramak".

Perhaps one way of having the README both displaying examples and redirect developpers and contributors to the doc is adding a "Contribute" section ?

Shimwell, 2020-09-24:
So we can add a Contribute section with a link to the github into the readme. 

Perhaps the readme could also include the diagram of all the components (just updated)
https://user-images.githubusercontent.com/8583900/94205189-a68f4200-feba-11ea-8c2d-789d1617ceea.png

Shimwell, 2020-12-16:
We can add the professional review to the docs and the readme [[PULLREQUEST] Paramak Project Review.pdf](https://github.com/ukaea/paramak/files/5704872/PULLREQUEST.Paramak.Project.Review.pdf)

Shimwell, 2021-01-27:
closing this as the examples are now in the read the docs

closed 2021-01-27 by Shimwell