ukaea/paramak #237: documenting properties

billingsley-john, 2020-09-09:
It is not convention to add docstrings to property setters - docstrings are only usually added to a property under @property. Sphinx ignores docstrings in property setters because of this. Some property setters have been documented but these docstrings do not appear in the documentation. This should be changed.

RemiTheWarrior, 2020-09-09:
@billingsley-john does it mean that properties shouldn't be documented at the class level anymore then ?

Shimwell, 2020-09-09:
In another project I've added doc strings to these attributes setters using the @property
https://github.com/ukaea/neutronics_material_maker/blob/72763629cfc9941d6aa5068fe6b304dde2a04c93/neutronics_material_maker/material.py#L235-L286

And it shows up in the docs
https://neutronics-material-maker.readthedocs.io/en/latest/material.html#neutronics_material_maker.material.Material.fispact_material

billingsley-john, 2020-09-09:
@Shimwell sorry, I realize I haven't explained correctly. Docstrings appear in the docs if they're added to the block starting with @Property, but they do not appear if they're added to the block starting with @points.setter (for example).

In shape.py, we have docstrings for points but they're added in @points.setter part rather than the @property part
https://github.com/ukaea/paramak/blob/develop/paramak/shape.py#L157-L168

billingsley-john, 2020-09-09:
> @billingsley-john does it mean that properties shouldn't be documented at the class level anymore then ?

I think we should still document properties/attributes, the docstrings just need to be in the correct place

closed 2020-09-17 by billingsley-john