ukaea/paramak #251: Varible offset for blanket

Shimwell, 2020-09-15:
To reproduce geometry relevant for EU DEMO designs the blankets need a little more flexibility.

In this case it looks like a variable offset from the plasma is needed to match the EU DEMO blanket shape

[image]

The segmentation of the blanket is looking better and an example has been added to the example_parametric_components branch. Eventually this can be part of a demo model

[image]
[image]

RemiTheWarrior, 2020-09-16:
@Shimwell the shape of the DEMO blanket you're showing kinda looks like the princeton D shape (with a straight end on the inner side). Are you sure that a variable offset would change that ?
Does tweaking the shaping parameters (triangularity and so on) help ?

Shimwell, 2020-10-26:
Variable offset implemented, I shall have another go at the demo blanket soon

closed 2020-10-26 by Shimwell