ukaea/paramak #253: parametric component points

billingsley-john, 2020-09-16:
Some parametric components still have last point=first point in their find_points() methods. These can all be removed.

billingsley-john, 2020-09-16:
This hasn't been picked up because the parametric_component classes have their own points setter, rather than inheriting the points setter from shape.py

billingsley-john, 2020-09-16:
Some parametric_components still require last point=first point as they overwrite the setter which automatically adds the final point

Shimwell, 2020-09-16:
Well spotted

billingsley-john, 2020-09-16:
The points getter and setter should not be overwritten in the parametric component classes. Instead, self.find_points() should be called immediately after the 'parametric parameters' (such as height, inner_radius etc) are set. I.e. self.find_points() does not have to be called as part of the points setter, meaning we can inherit the points getter/setter pair instead and make use of the points checks defined in shape.py

billingsley-john, 2020-09-17:
Resolved in #254

closed 2020-09-17 by billingsley-john