ukaea/paramak #261: cut intersect union

billingsley-john, 2020-09-17:
cut, intersect and union are shared methods for each class. Therefore, it isn't necessary to have getters/setters for each method in each shape class - these should be placed in shape.py and inherited

billingsley-john, 2020-09-17:
However, cut, intersect and union are not used directly by the Shape class. They are used by the parametric_shape classes in the create_solid() method. Moving cut, intersect and union to Shape would reduce the number of lines of code, but add properties to the Shape class that cannot be used directly by the Shape class.
What do you think @Shimwell, @RemiTheWarrior ?

RemiTheWarrior, 2020-09-17:
I find it strange to have a method/property belonging to a class that cannot use it directly. On the other hand it'd be great for refactoring. Is there a way `Shape()` can actually use it ?

billingsley-john, 2020-09-17:
I suppose the boolean methods could be put into a method in shape.py, allowing cut, intersect and union to be used by Shape. These methods could then be inherited by parametric_shapes and used as part of the create_solid() method. This is maybe something for the future then as this is a large code refactoring task.

billingsley-john, 2020-09-21:
Resolved in #263

closed 2020-09-21 by billingsley-john