ukaea/paramak #273: InnerTfCoilsFlat ignores user azimuth_placement_angle arg

Shimwell, 2020-09-21:
When making a InnerTfCoilsFlat and including the optional azimuth_placement_angle it is ignored by the code

If a user azimuth_placement_angle is provided perhaps this should overrule the internal calculation of azimuth_placement_angle.

Alternativly one could allow the user to supply an angular offset

```
    inner_tf = paramak.InnerTfCoilsFlat(
        height=1800,
        inner_radius=330,
        outer_radius=430,
        number_of_coils=number_of_toroidal_field_coils,
        gap_size=5,
        azimuth_placement_angle=np.linspace(
            0, 360, 16,
            endpoint=False
            )
    )
```

Shimwell, 2020-09-21:
ToroidalFieldCoilPrincetonD also ignores this user arg

billingsley-john, 2020-09-22:
Possible to solve this either using an angular offset specifying the start angle for azimuthal placement, or allow control of the azimuth_placement_angle parameter.
However, if we allow users to use the azimuth_placement_angle parameter, this can become confusing if they also try to use the number_of_coils parameter (could get a difference in the number of coils produced).

billingsley-john, 2020-09-29:
Resolved in #307 using an offset angle.
azimuth_placement_angle cannot be passed by the user because it is calculated internally by the class.
Users can define an azimuth_start_angle to specify the angle of the center of the first tf coil - the coils are then equally spaced around 360 degrees

closed 2020-09-29 by billingsley-john