ukaea/paramak #274: Helper points on ToroidalFieldCoilPrincetonD

Shimwell, 2020-09-21:
When making a ToroidalFieldCoilPrincetonD it makes a nice outer TF coil hoop. 

Users must then make an inner tf coil leg. It would therefore be useful if the coordinates of the points where inner tf leg connects are more accessible from the ToroidalFieldCoilPrincetonD

If the coordinates of the green patch on this diagram were available as attributes on the instance of the class than that would be great

Something like this 

```
my_outer_tf = ToroidalFieldCoilPrincetonD(random args)
print(my_outer_tf.inner_leg_connection_points)
> [(300, 400), (300, -400), (350, -400), (350, 400)]
```

[image]

RemiTheWarrior, 2020-09-22:
@Shimwell I think this is already given for the Triple Arc magnet. I shall try and have a look a the PrincetonD class

Shimwell, 2020-09-23:
Thanks for fixing this Remi

closed 2020-09-23 by Shimwell