ukaea/paramak #280: rotation_angle not working for InnerTfCoilsFlat

Shimwell, 2020-09-22:
Normally the rotation_angle reduces the rotated geometry in size from 360 to 0

However for some solids that are not rotated (e.g extruded) the rotation angle doesn't do much.

InnerTfCoilsFlat, InnerTfCoilsCircle, ToroidalFieldCoilRectangle are just some of the shapes where the rotation_angle has no impact. It would be great if rotation_angle also impacted these shapes.

One example of where this would save code is in the submersion_reactor.py and ball_reactor script where a cutting_slice is introduced to effectively act as the rotation_agnle limits and trim the size of the tf coils.

However we could alternativly add create_solid methods to the classes and perform a cut with a cutting slice within each of the component classes instead.

perhaps there is some link up with #273

```
cutting_slice = paramak.RotateStraightShape(
                points=[
                    (0, max_high),
                    (max_width, max_high),
                    (max_width, -max_high),
                    (0, -max_high),
                ],
                rotation_angle=360 - self.rotation_angle,
                azimuth_placement_angle=360 - self.rotation_angle,
            )
```

billingsley-john, 2020-10-05:
New parametric shapes, CuttingSlice and CuttingSliceFP have been added which can be utilised to perform this function. CuttingSliceFP takes the rotation_angle of the shape passed to it and performs the according cut. The rotation_angle parameter needs to be added to the extruded shapes so it can be accessed by CuttingSliceFP and the appropriate cut can be made. This then needs to be made automatic in the extruded parametric components, such as magnets

RemDelaporteMathurin, 2020-11-17:
@billingsley-john can this be closed ?

closed 2020-11-24 by RemDelaporteMathurin