ukaea/paramak #283: BlanketFP offset_from_plasma not working when plasma not provided

Shimwell, 2020-09-23:
when using the BlanketFP with an offset_from_plasma it is sometimes ignored

When providing plasma=plasma to the BlanketFP then the offset works

However when providing the same plasma in terms of minor radius elongation and triangulation it appears to ignore the offset from plasma argument

Ignores offset
```
blanket = paramak.BlanketFP(
    minor_radius=150.0,
    major_radius=450.0,
    triangularity=0.55,
    elongation=2.0,
    thickness=100,
    offset_from_plasma=25,
    start_angle=-160,
    stop_angle=160,
    rotation_angle=180
)
```

[image]

while this appears to take the offset_from_plasma into account
```
plasma = paramak.Plasma(
             minor_radius=150.0,
             major_radius=450.0,
             triangularity=0.55,
             elongation=2.0,
             rotation_angle=180
)

blanket = paramak.BlanketFP(
    plasma=plasma
    thickness=100,
    offset_from_plasma=25,
    start_angle=-160,
    stop_angle=160,
    rotation_angle=180
)
```
[image]

RemiTheWarrior, 2020-09-23:
I'm on it

Shimwell, 2020-09-23:
Thanks for fixing this, I've tested and can confirm it works as hoped

closed 2020-09-23 by Shimwell