ukaea/paramak #290: feature request for BlanketFP to accept tuple of more than two values

Shimwell, 2020-09-23:
The recently improved BlanketFP allows for offset_from_plasma to be specified as a tuple of two values. 

This is then sets the offset for the start and stop angle and interpolates between these values.

When making a blanket from a combination of a radial and vertical build it is common to have 4 offsets, these are upper, lower, inner, outer.

So I'm wondering if the offset and thickness could be future enhanced to accept a tuple or list of more than two values and then interpolate between the list and the start stop angles

I think it is common to add offsets for 0, 90, 180 and 270 degrees when building from radial and vertical build information

Perhaps this is tricky to implement and the better way to do it would be for the user to make a callable function

RemiTheWarrior, 2020-09-25:
@Shimwell I see what you mean. I can take a look but I think, as you said, make a callable function which interpolates between four offsets is way more simple :smile: 

If this is really unconvenient I'll try and write a function :smile:

Shimwell, 2020-09-25:
I am happy writing the callable function but I'm not sure if all the users are, could we embed a callable function for linear interpolation inside the BlanketFP

Shimwell, 2020-09-29:
Thanks Remi for solving this in #301

closed 2020-09-29 by Shimwell