ukaea/paramak #310: Diagram with labels for ToroidalFieldCoilPrincetonD

Shimwell, 2020-09-28:
Adding images to ToroidalFieldCoilPrincetonD

We could do with some labels to explain R1 and R2 parameters in the [Mathcha](https://www.mathcha.io/editor) diagrams 

I have started the diagrams but am not quite sure where R1 and R2 go, any hints @RemiTheWarrior 

Also I added a create_solid method to the ToroidalFieldCoilPrincetonD so that the innerleg can be made, hence the diagram includes the inner leg

[image]
[image]

RemiTheWarrior, 2020-09-29:
[image]
I apologise for the poor quality of this diagram 😄

Shimwell, 2020-09-29:
[image]

Thanks for the hints @RemiTheWarrior I shall update the docs soon

RemiTheWarrior, 2020-09-29:
@Shimwell you're very welcome. What about placing the origin as well ?

closed 2020-10-19 by Shimwell