ukaea/paramak #312: setting workplane after construction is being ignored

Shimwell, 2020-09-29:
I came across this bug while making images for the docs

It appears the svg export breaks when 3 or more magnets are used

```
import paramak

component = paramak.ToroidalFieldCoilTripleArc(
    R1=80,
    h=200,
    radii=(70, 100),
    coverages=(60, 60),
    thickness=30,
    distance=30,
    number_of_coils=1,
    workplane='XY'
)
component.export_svg('1_magnets.svg')
```

[image]

This appears to not work
```
component = paramak.ToroidalFieldCoilTripleArc(
    R1=80,
    h=200,
    radii=(70, 100),
    coverages=(60, 60),
    thickness=30,
    distance=30,
    number_of_coils=3,
    component.workplane = "XY"
)
component.export_svg('3_magnets.svg')
```

[image]

billingsley-john, 2020-09-29:
I think this is linked to #308

RemDelaporteMathurin, 2020-11-02:
@Shimwell i think the title of this issue is incorrect. The workplane _can_ be set after construction (see below).

```python
component = paramak.ToroidalFieldCoilTripleArc(
    R1=80,
    h=200,
    radii=(70, 100),
    coverages=(60, 60),
    thickness=30,
    distance=30,
    number_of_coils=3,
)
```
produces : 
[image]

Which is the expected behaviour.

```python
component = paramak.ToroidalFieldCoilTripleArc(
    R1=80,
    h=200,
    radii=(70, 100),
    coverages=(60, 60),
    thickness=30,
    distance=30,
    number_of_coils=3,
)
component.workplane = "XY"
```

produces 
[image]

Which also is the expected since the rotation axis in Shape.rotate_solid() is independent on workplane.
Regardless of the workplane, the rotation axis will always be the Z axis : 
https://github.com/ukaea/paramak/blob/10552f1b89820dd0f7a08e4a126834877e3106b4/paramak/shape.py#L514-L517

One way to fix this would be to make the rotation axis depend on the workplane.
I think it can even be a parameter that, by default, makes the rotation axis orthogonal to the workplane. It can be useful to have flexibility here.

What do you think @Shimwell @billingsley-john  ?

Shimwell, 2020-11-02:
this is perfect

RemDelaporteMathurin, 2020-11-11:
@Shimwell shall we close this ?

Shimwell, 2020-11-11:
Might be related to this one as well #462

PR #428 (2020-11-02): Variable rotation axis for azimuth_placement_angle
## Proposed changes

This PR fixes #312 by making the rotation axis used in `shape.rotate_solid()` variable.
To this end, a new argument/attribute has been added as rotation_axis.
rotation_axis accepts a string in `["X", "Y", "Z", "-X", "-Y", "-Z", "+X", "+Y", "+Z"]`, a list of two points or `None`.

**Usage**
Default behaviour
```python
import paramak

component = paramak.ToroidalFieldCoilTripleArc(
    R1=80,
    h=200,
    radii=(70, 100),
    coverages=(60, 60),
    thickness=30,
    distance=30,
    number_of_coils=3,
    workplane="YZ"
) [... cut]
closed 2020-11-17 by RemDelaporteMathurin