ukaea/paramak #338: construction errors with CenterColumnStudyReactor

Shimwell, 2020-10-06:
With some arguments settings the CenterColumnStudyReactor fails to produce geometry

The following model doesn't work over here

```
import paramak            
my_reactor = paramak.CenterColumnStudyReactor(
    inner_bore_radial_thickness=50,
    inboard_tf_leg_radial_thickness=100,
    center_column_shield_radial_thickness_mid=50,
    center_column_shield_radial_thickness_upper=70,
    inboard_firstwall_radial_thickness=20,
    divertor_radial_thickness=50,
    inner_plasma_gap_radial_thickness=50,
    plasma_radial_thickness=100,
    outer_plasma_gap_radial_thickness=50,
    center_column_arc_vertical_thickness=200,
    plasma_high_point=(350, 0),
    plasma_gap_vertical_thickness=20,
    rotation_angle=360,
)
```

Perhaps this is a good time to rewrite the model so that is avoids the use of plasma_high_point which can be difficult for users to set correctly

Shimwell, 2020-10-24:
Getting some more construction errors randomly on the tests
https://app.circleci.com/pipelines/github/ukaea/paramak/1091/workflows/da9b52c2-c283-4302-bd20-05c23d2fe8dc/jobs/1123

RemDelaporteMathurin, 2020-11-11:
Is this related to #445 ?

Shimwell, 2020-11-12:
I think I have figured this one out and have a fix in the works. Basically the plasma_high_point is a tricky way to go about building a reactor

RemDelaporteMathurin, 2020-11-17:
@Shimwell this has been fixed in #470. Closing this

PR #470 (2020-11-12): Switching to elongation triangularity instead of plasma high point
## Proposed changes

This fixes #338 by redesigning the submersion and study reactors to use plasma elongation and plasma triangular instead of plasma high point. This is nice as plasma high point was always tricky for users to set as it needed to be made at the correct x value (not too big and not too small).  However it is also a breaking change as the submersion reactor and the center column study reactor now accept different input arguments. Tests have bee updated accordingly

[image]

## Types of changes

What types of cha [... cut]
closed 2020-11-17 by RemDelaporteMathurin