ukaea/paramak #345: adding checks to PoloidalFieldCoilFP corner_points setter

Shimwell, 2020-10-09:
Nice and simple issue suitable for a first contribution

The setter for the corner_points part of the PoloidalFieldCoilFP class should check if the value passed is the correct type and it it is not an error should be raised

These are the lines that need fixing

https://github.com/ukaea/paramak/blob/12b75aa2a2bee7c589b0e3ff7fce2fd9dc0d77e8/paramak/parametric_components/poloidal_field_coil_fp.py#L97-L101

This is how to raise an error
```raise ValueError("descriptive message goes here")```

Here is an example of a setter that checks if the value is the correct type 
https://github.com/ukaea/paramak/blob/12b75aa2a2bee7c589b0e3ff7fce2fd9dc0d77e8/paramak/reactor.py#L87-L91

TediaN97, 2020-10-09:
i would like if u assign me on this issue

somstrom, 2020-10-09:
@Shimwell 

Could you please assign this to me ?

Shimwell, 2020-10-12:
Hi @somstrom and @TediaN97 

Thanks for both volunteering to fix this one, help is always appreachiated

@TediaN97 I should have assigned you to this as you were first but somehow I only noticed when a scrolled up a bit.

I've learned my lesson and won't use my phone for such things in the future :-)

I think I have another very very similar issue that I can write up. Let me made up that issue then assign if you are still keen. Sorry for the mix up.

Shimwell, 2020-11-02:
Thanks everyone for taking a look at this, it has now been fixed by #421

closed 2020-11-02 by Shimwell