ukaea/paramak #346: Adding extrude both as an options

Shimwell, 2020-10-09:
This is a nice and easy issue ideal for a first contribution 

The ExtrudeMixedShape and the ExtrudeSplineShape could do with an optional argument that allow extrusion in one or both directions.

Both directions is currently hard coded in these two classes

These two files that will need changing

https://github.com/ukaea/paramak/blob/develop/paramak/parametric_shapes/extruded_spline_shape.py
https://github.com/ukaea/paramak/blob/develop/paramak/parametric_shapes/extruded_mixed_shape.py

There are two other extrude shapes that already have this feature so might be useful (ExtrudeStraightShape and ExtrudeCirlceShape)

Here are the relevant lines from ExtrudeStraightShape that can be copied

https://github.com/ukaea/paramak/blob/12b75aa2a2bee7c589b0e3ff7fce2fd9dc0d77e8/paramak/parametric_shapes/extruded_straight_shape.py#L21
https://github.com/ukaea/paramak/blob/12b75aa2a2bee7c589b0e3ff7fce2fd9dc0d77e8/paramak/parametric_shapes/extruded_straight_shape.py#L42
https://github.com/ukaea/paramak/blob/12b75aa2a2bee7c589b0e3ff7fce2fd9dc0d77e8/paramak/parametric_shapes/extruded_straight_shape.py#L82
https://github.com/ukaea/paramak/blob/12b75aa2a2bee7c589b0e3ff7fce2fd9dc0d77e8/paramak/parametric_shapes/extruded_straight_shape.py#L105

somstrom, 2020-10-09:
@Shimwell 

Could you please assign this to me ?

TediaN97, 2020-10-09:
i would like if u assign me on this issue

Shimwell, 2020-10-12:
Thanks for volunteering again both of you. Sorry for my slow response. Looking forward to seeing what can be done here

Shimwell, 2020-10-22:
This might have been solved by PR #387

RemDelaporteMathurin, 2020-10-26:
Fixed in #393

PR #387 (2020-10-23): Parametric shapes refactoring
## Proposed changes

This PR proposes a refactoring of the parametric shapes.
Instead of having parametric shapes inherit from shape, the non-mixed shapes (eg. `ExtrudeStraightShape`, `RotateSplineShape`) now inherit from the mixed shape (eg, `ExtrudeMixedShape`, `RotateMixedShape`).

This allows to avoid code repetition (`create_solid` method).

`RotateCircleShape` and `ExtrudeCircleShape` aren't impacted by this refactoring since these classes aren't used in the same way.

Others:
- Fixes #346 

## Types of changes

What types of changes does your code introduce [... cut]
PR #393 (2020-10-26): Parametric shape refactoring
## Proposed changes

This is the same as the previous PR from Remi but just done on the internal branches

This PR proposes a refactoring of the parametric shapes.
Instead of having parametric shapes inherit from shape, the non-mixed shapes (eg. `ExtrudeStraightShape`, `RotateSplineShape`) now inherit from the mixed shape (eg, `ExtrudeMixedShape`, `RotateMixedShape`).

This allows to avoid code repetition (`create_solid` method).

`RotateCircleShape` and `ExtrudeCircleShape` aren't impacted by this refactoring since these classes aren't used in the same way.

Other [... cut]