ukaea/paramak #353: BUG: PortCutterRotated with polar_coverage_angle > 180

RemDelaporteMathurin, 2020-10-11:
When using the PortCutterRotated component, polar_coverage_angle attribute cannot be greater than 180 and a Standard_ConstructionError is raised.

This might be due to lines below:
https://github.com/ukaea/paramak/blob/7f5acbd65f904a1ccb527e061a3b7e3f86b8a26f/paramak/parametric_components/port_cutters.py#L151-L163

**How to reproduce**

```python
from paramak import PortCutterRotated

shape = PortCutterRotated(
    center_point=(0, 0),
    polar_coverage_angle=180,
    polar_placement_angle=0,
    max_distance_from_center=30,
    rotation_angle=10

)
shape.solid
```

Shimwell, 2020-10-13:
Ah yes good spot, perhaps for this one we can just check the inputs and limit them to below 180.

As I don't think a port opening above 180 degrees is ever needed

RemDelaporteMathurin, 2020-10-13:
Perhaps. In any case the lines below need to be tested I think. Also I didn't get why the outer points couldn't be negative ?

https://github.com/ukaea/paramak/blob/7f5acbd65f904a1ccb527e061a3b7e3f86b8a26f/paramak/parametric_components/port_cutters.py#L151-L163

RemDelaporteMathurin, 2020-10-19:
@Shimwell perhaps this could be a "good first issue" bug