ukaea/paramak #363: ading getter and setter to check input for the BallReactor

Shimwell, 2020-10-15:
The BallReactor class https://github.com/ukaea/paramak/blob/main/paramak/parametric_reactors/ball_reactor.py accepts some arguments from the user including:

- pf_coil_radial_thicknesses
- pf_coil_vertical_thicknesses

These two arguments should be a list and the type list should be checked on creation using a setter method like we have in other areas e.g. https://github.com/ukaea/paramak/blob/7f5acbd65f904a1ccb527e061a3b7e3f86b8a26f/paramak/shape.py#L229-L236

A getter will also need adding for each

This is nice and easy addition

shen3443, 2020-10-15:
Is this still available? I'd like to help out if it is

Shimwell, 2020-10-16:
Yep this is still available any help is welcome

Shimwell, 2020-10-16:
Solved by @shen3443 , thanks for this contribution

closed 2020-10-16 by Shimwell