ukaea/paramak #368: Getting args for ITERtypeDivertorNoDome in docs

Shimwell, 2020-10-16:
The args for the ITERtypeDivertorNoDome in the docs are not showing up, but perhaps this is ok as it inherits.

https://paramak.readthedocs.io/en/latest/paramak.parametric_components.html#paramak.parametric_components.divertor_ITER_no_dome.ITERtypeDivertorNoDome

RemDelaporteMathurin, 2020-10-16:
@Shimwell this is because in fact there isn't any argument specific to ITERtypeDivertorNoDome that doesn't live in ITERtypeDivertor

RemDelaporteMathurin, 2020-10-17:
True. In the Dome version, if dome is False, then all these parameters (dome_height, dome_pos, dome_thickness) are ignored.

Perhaps the noDome version shouldn't exist to avoid confusion. The two images (dome and nodome) could illustrate the same ITERtypeDivertor class.
What do you think ?

Shimwell, 2020-10-19:
I'm wondering if we have done it the wrong way around, perhaps we should inherit and extend instead of inherit and suppress

RemDelaporteMathurin, 2020-10-20:
In my opinion it is a lot easier to inherit and remove since the method that creates the dome points take other points as argument

closed 2020-11-23 by RemDelaporteMathurin