ukaea/paramak #392: Testing for extrude_both

billingsley-john, 2020-10-23:
Simple tests need to be added to test the extrude_both function.
extrude_both=True = volume
extrude_both=False = volume * 0.5

RemDelaporteMathurin, 2020-10-23:
I can maybe add them to #393

Shimwell, 2020-10-23:
I saw this one which is similar as it checks for 0.5 of the volume.

https://github.com/ukaea/paramak/blob/b3ff4d99f737ee8ca4fea4866fdda18417204817/tests/test_parametric_shapes/test_ExtrudeStraightShape.py#L128-L141

But I am not sure if both actually changes the volume. I think both just shares the distance in two directions instead of one. If you want to check the volume you might have to get creative with some Boolean cuts

Shimwell, 2020-10-23:
Ah yes sorry, you are right, both does certainly double the distance. Perhaps we need to update a few things then as I think our docstrings imply that the distance setting is the total extrusion distance

Shimwell, 2020-10-23:
This code snippet might be hand for the tests

```
import paramak

cube1 = paramak.ExtrudeStraightShape(points=[(10, 10), (20, 10), (20, 20), (10,20)],
                                    distance=120,
                                    extrude_both=False,
                                    material_tag='cub_mat',
                                   azimuth_placement_angle=10)
cube2 = paramak.ExtrudeStraightShape(points=[(10, 40), (20, 40), (20, 30), (10,30)],
                                    distance=120,
                                    extrude_both=True,
                                    material_tag='cub_mat',
                                   azimuth_placement_angle=10)
print(cube1.volume)
print(cube2.volume)
my_reactor = paramak.Reactor(shapes_and_components=[cube1,cube2])
my_reactor.solid
```
[image]

RemDelaporteMathurin, 2020-10-23:
I've added tests to #393

RemDelaporteMathurin, 2020-10-26:
Tests have been added in #393

closed 2020-10-26 by RemDelaporteMathurin