ukaea/paramak #409: Random issue : test_adding_multiple_shapes_with_material_tag_to_reactor

RemDelaporteMathurin, 2020-10-27:
This test randomly fails. I think this is due to this line which seems to randomise the order of the `.material_tags` attribute.

https://github.com/ukaea/paramak/blob/23d50e9a81c1be846016dba106c8f6590308f78e/paramak/reactor.py#L76

In this minimal working example, `count` is sometimes 0, sometimes 500.
```python
count = 0
for i in range(500):
    a = list(set(['mat1', 'mat2']))
    if a[0] != "mat1":
        count += 1
print(count)
```

I don't understand why though...

RemDelaporteMathurin, 2020-10-27:
https://stackoverflow.com/questions/9792664/converting-a-list-to-a-set-changes-element-order

RemDelaporteMathurin, 2020-10-27:
@Shimwell is `set()` really needed here ?

Shimwell, 2020-10-27:
Ah I see. So I was using ``set`` to avoid duplicate entries that are the same.

But I understand that set does not preserve the order.

Perhaps the use of ```set``` can be changed to something like this ```[x for x in a if x not in b]```

Shimwell, 2020-10-27:
Or we can change the test to allow any order within the list

RemDelaporteMathurin, 2020-10-28:
In #405 I removed the list(set()) transform since values is already a list

https://github.com/ukaea/paramak/blob/646cab369e1a19e6019414d0caa6f3caff53bada/paramak/reactor.py#L65-L76

RemDelaporteMathurin, 2020-10-30:
Fixed in #421

PR #405 (2020-10-28): Create solid refactoring
## Proposed changes

This PR proposes a refactoring of the `create_solid `methods for all the parametric shapes.
All the code repetition has been put in `Shape().create_solid()` method (which makes it kinda big now btw so maybe slicing it into independant functions would be the way to go one day).

Also, as in #387, the classes SweepSplineShape and SweepStraightShape inherit from SweepMixedShape for simplification purposes.

This greatly reduces code repetition.

Other : fixes #409 

## Types of changes

What types of changes does your code introduce to the Paramak?
_P [... cut]
closed 2020-10-30 by RemDelaporteMathurin