ukaea/paramak #410: rotation_angle for extruded shapes

billingsley-john, 2020-10-27:
rotation_angle < 360 degrees causes a cutting wedge shape to be appended to shape.cut. Each time rotation_angle is specified, a different cutting wedge is appended to shape.cut, causing all previous wedge cut shapes to remain, which is wrong. Instead, each new rotation_angle specification should replace the previous cutting wedge shape in shape.cut.

RemDelaporteMathurin, 2020-10-28:
With @Shimwell we noticed that tests like below can be applied to instances. Maybe that can solves this.

https://github.com/ukaea/paramak/blob/2a197b383650f7f8d4d01734bce8f7bfda377071/paramak/parametric_components/blanket_cutter_parallels.py#L78-L84

billingsley-john, 2020-10-28:
Resolved in #419

closed 2020-10-28 by billingsley-john