ukaea/paramak #445: RotateSplineShape() volume = 0 if rotation_angle=360

billingsley-john, 2020-11-06:
As described

RemDelaporteMathurin, 2020-11-08:
If that can help, the following MWE reproduces the issue by creating an empty STL file:

```python
import paramak
points = [
    (1, 0),
    (2, 0),
    (2, 1),
    (1, 1)
]
shape = paramak.RotateSplineShape(points=points)
shape.export_stl("out.stl")
```

Shimwell, 2020-11-08:
Perhaps we should update the cadquery version we are running

billingsley-john, 2020-11-13:
RotateCircleShape is also having the same issue

billingsley-john, 2020-11-13:
Also the same issue for .area and .areas

billingsley-john, 2020-11-13:
These issues are not necessarily due to a failed solid construction. In some cases, a solid is correctly produce but the area and volume attributes do not return the expected values.

RemDelaporteMathurin, 2020-11-13:
With the MWE I put up above, it creates a solid but the STP file is empty (0 Ko)

RemDelaporteMathurin, 2020-11-14:
@billingsley-john @Shimwell here's a MWE for testing/debugging: 
```python
import paramak
import numpy as np

# create a circle distribution
thetas = np.linspace(0, 2*np.pi, 100, endpoint=False)
offset = 1
radius = 1
x, y = offset + radius*(np.cos(thetas) + 1), radius*np.sin(thetas)

points = [[X, Y] for X, Y in zip(x, y)]

# create 3 nearly identical shapes
shape_spline = paramak.RotateSplineShape(
    points=points[:])

shape_circle = paramak.RotateCircleShape(
    radius=radius, points=[[2, 0]])

shape_straight = paramak.RotateStraightShape(
    points=points[:])

# print volumes
print(shape_spline.volume)
print(shape_circle.volume)
print(shape_straight.volume)

# export to STL
shape_circle.export_stl('circle.stl')
shape_spline.export_stl('spline.stl')
shape_straight.export_stl('straight.stl')

```

RemDelaporteMathurin, 2020-11-14:
I've narrowed it down to 2 problems:

1.  revolve has to be used with clean=False argument (see [this cadquery issue](https://github.com/CadQuery/cadquery/issues/322))
2. the rotate_solid method breaks stuff ... see the MWE below

```python
import cadquery as cq

## works
solid = cq.Workplane("XY").moveTo(30, 30).circle(10).revolve(360, clean=False)
print(solid.val().Volume())  # prints 59217.62640653616

## doesn't work
solid = cq.Workplane("XY").moveTo(30, 30).circle(10).revolve(360)
print(solid.val().Volume())  # prints 0.0

## also doesn't work
solid = cq.Workplane("XY").moveTo(30, 30).circle(10).revolve(360, clean=False)

# this mimics the paramak.Shape().rotate_solid method
solid.rotate((0, 0, -1), (0, 0, 1), 0)
solid = cq.Workplane("XY").union(solid)

print(solid.val().Volume())  # prints 0.0
```

1. can be fixed by maybe upgrading the cadquery version as @Shimwell proposed
2. can be fixed by adding a test checking if self.rotation_angle is not 360: but that's not ideal

RemDelaporteMathurin, 2020-11-14:
Update: the previous MWE I posted has the expected behaviour when run with the latestest cadquery version. Updating cadquery should solve this ! 🥳

closed 2020-11-20 by RemDelaporteMathurin