ukaea/paramak #462: InnerTfCoilsFlat being make with hole

Shimwell, 2020-11-11:
While running the [magnet set](https://github.com/ukaea/paramak/blob/main/examples/example_parametric_components/make_magnet_set.py) example i noticed that the InnerTfCoilsFlat was being made with a hole in it. This is probably something to do with the work plane changes but I'm not really sure. These should be continuous straight sections but something strange is happening here is the code to reproduce.

```python
import paramak

inner_tf_case = paramak.InnerTfCoilsFlat(
    height=1800,
    inner_radius=330,
    outer_radius=430,
    number_of_coils=8,
    gap_size=5,
    rotation_angle=180,
    azimuth_start_angle=(360 / 8) / 2.
)

inner_tf_case.solid
```

[image]
[image]

RemDelaporteMathurin, 2020-11-11:
Have I messed something up with #428 ?

RemDelaporteMathurin, 2020-11-11:
@Shimwell I am not sure it has to do with workplane but rather cutting wedges.

Removing the rotation_angle argument doesn't produce this bug : 

```python
import paramak

inner_tf_case = paramak.InnerTfCoilsFlat(
    height=1800,
    inner_radius=330,
    outer_radius=430,
    number_of_coils=8,
    gap_size=5,
    # rotation_angle=180,
)
inner_tf_case.solid
```
[image]

RemDelaporteMathurin, 2020-11-11:
@Shimwell it's confirmed:
```
import paramak

inner_tf_case = paramak.InnerTfCoilsFlat(
    height=1800,
    inner_radius=330,
    outer_radius=430,
    number_of_coils=8,
    gap_size=5,
    rotation_angle=180,
)
inner_tf_case.solid
inner_tf_case.export_stl('out.stl')

cutting_wedge = paramak.CuttingWedgeFS(inner_tf_case)
cutting_wedge.export_stl('cutting_wedge.stl')
```
[image]
[image]

RemDelaporteMathurin, 2020-11-11:
Problem is CuttingWedgeFS uses the shape workplane (which is in this case the XY plane).
I think the CuttingWedge workplane should be computed from the rotation axis rather than from the shape workplane.

The following allows to have the correct orientation, but the dimensions aren't correct.

```python
    def get_workplane(self):
        workplanes = ["XY", "YZ", "XZ"]
        for wp in workplanes:
            if self.rotation_axis in wp:
                return wp
```

Shimwell, 2020-11-11:
Ah great detective work

PR #464 (2020-11-12): Fix cutting wedge with different workplanes
## Proposed changes

Fixes #462 

```python
import paramak

number_of_toroidal_field_coils = 8
angle_offset = (360 / number_of_toroidal_field_coils) / 2.

inner_tf_case = paramak.InnerTfCoilsFlat(
    height=1800,
    inner_radius=330,
    outer_radius=430,
    number_of_coils=number_of_toroidal_field_coils,
    gap_size=5,
    rotation_angle=180,
    azimuth_start_angle=angle_offset
)

pf_coils = paramak.PoloidalFieldCoilSet(
    heights=[100, 120, 80, 80, 120, 180],
    widths=[100, 120, 80, 80, 120, 180],
    center_points=[
        (530, 1030),
  [... cut]
closed 2020-11-12 by RemDelaporteMathurin