ukaea/paramak #493: Rotate Shapes understanding of azimuth_placement_angle results in misaligned ports

Shimwell, 2020-11-14:
While making the vacuum vessel for #492 I noticed that the rotated port cutters do not line up with the circle port cutters or the rectangle port cutters.

They all have the same azimuth_placement_angle but the ports appear to have different rotations.  

The circle port cutters or the rectangle port cutters are centered around the azimuth_placement_angle.
The rotated port cutter states at the azimuth_placement_angle and rotates and attional degrees

So perhaps this relates to something more fundamental. Should the rotate shapes use the azimuth_placement_angle to position their centers or start from those angles. 

It might be tricky to get the cutters components to line up unless this gets changed

[image]

RemDelaporteMathurin, 2020-11-14:
The rotated ports and the circular ports seem to be 90° out of phase. I think I added something to modify the azimuth placement angle but we removed it from the PR for some reason.

```python
import paramak

angles = [0, 45, 90]

vv = paramak.VacuumVessel(10,  3, 0.2, rotation_angle=180)
port_circular = paramak.PortCutterCircular(0, 0.5, 3, azimuth_placement_angle=angles)
port_rotated = paramak.PortCutterRotated(
        center_point=(0, 0),
        polar_coverage_angle=10,
        rotation_angle=10,
        polar_placement_angle=25,
        max_distance_from_center=3,
        azimuth_placement_angle=angles
    )

vv.export_stl('vv.stl')
port_circular.export_stl("circular.stl")
port_rotated.export_stl("rotated.stl")
```

RemDelaporteMathurin, 2020-11-14:
An easy workaround : 

```python
import paramak
import numpy as np

angles = np.linspace(0, 360, 12)

port_circular = paramak.PortCutterCircular(0, 0.5, 10)

# til 90deg dephasing is fixed
port_circular.azimuth_placement_angle = (angles-90).tolist()

rotation_angle = 10
port_rotated = paramak.PortCutterRotated(
        center_point=(0, 0),
        polar_coverage_angle=rotation_angle,
        rotation_angle=rotation_angle,
        polar_placement_angle=25,
    )
# change the angles
port_rotated.azimuth_placement_angle = (angles-rotation_angle/2).tolist()

vv = paramak.VacuumVessel(
    5,  3, 0.2,
    rotation_angle=180,
    cut=[port_rotated, port_circular])

vv.export_stl('vv.stl')
```
[image]

This modification of azimuth_placement_angle could be done internally.

Shimwell, 2020-11-15:
Great stuff Remi, which solution do you prefer. They both look good to me.

We can add the 90 offset into the components the need it or add that -90 part to the example

RemDelaporteMathurin, 2020-11-16:
If you want we can add these "angle offsets" to both these ports

Shimwell, 2020-11-16:
Ok that is the plan then. Will merging in 492 in the meantime

billingsley-john, 2020-11-20:
The default workplane for extruded shapes is "XZ". As the workplane is the plane in which the face of the shape is defined, extrusions from this workplane are in the Y direction. As the points of each port cutter are (0, Z), a port placed at azimuth_placement_angle = 0 in the "XZ" workplane projects along the Y axis. This makes it look like it is offset by 90 degrees from the correct axis but is not a bug - they are working as they are being told. To solve this, the workplane and rotation_axis of the ports simply need to be changed to "ZY" and "Z" respectively.

billingsley-john, 2020-11-20:
workplane = "ZY"
rotation_axis = "Z"

[image]
[image]

RemDelaporteMathurin, 2020-11-20:
@billingsley-john seems like you found a nice way of dealing with it !

closed 2020-11-24 by RemDelaporteMathurin