ukaea/paramak #494: Can't build image on Windows

RemDelaporteMathurin, 2020-11-14:
Running `docker build .` on Windows throws an error:

```
#7 3.354 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
------
failed to solve with frontend dockerfile.v0: failed to build LLB: executor failed running [/bin/sh -c apt-get install -y libgl1-mesa-glx]: runc did not terminate sucessfully
```

Adding `apt-get update` to the line below seems to solve it.
https://github.com/ukaea/paramak/blob/87008817a9f89d3bf0308f79f1b14e8fc878502f/Dockerfile#L18

Shimwell, 2020-11-15:
I think the dockerfile was updated recently. This new one might work better https://github.com/ukaea/paramak/blob/develop/Dockerfile

But I think you can also download the image like this

docker pull ukaea/paramak

And run it like this for terminal access
sudo docker run -it paramak
Or like this for a jupyter notebook
docker run -p 8888:8888 paramak

Shimwell, 2020-11-15:
ah perhaps I need to change the tag name a bit for the docker pull to work. Will do so soon

Shimwell, 2020-11-16:
#501 contains so details for the readme and the installation section of the docs

RemDelaporteMathurin, 2020-11-16:
> But I think you can also download the image like this
> 
> `docker pull ukaea/paramak:dev`
> 
> And run it like this for terminal access
> `docker run -it ukaea/paramak:dev`
> Or like this for a jupyter notebook
> `docker run -p 8888:8888 ukaea/paramak:dev`

The idea was to see if #445 could be fixed by using the nightly version of cadquery. Therefore, neither the ukaea/paramak nor the openmcworkshop/paramak could be used since they both use cadquery=2 version.

Shimwell, 2020-11-16:
Ah got you, ok I can change both images to use cadquery master

RemDelaporteMathurin, 2020-11-16:
Building with the current Dockerfile on Windows raises:
```
failed to solve with frontend dockerfile.v0: failed to build LLB: executor failed running [/bin/sh -c conda install -c conda-forge -c cadquery cadquery=master &&     conda install jupyter -y --quiet &&     conda clean -afy]: runc did not terminate sucessfully
```

Shimwell, 2020-11-16:
Ah my fault, I can fix this

Shimwell, 2020-11-25:
I think this might be fixed with the latest dockerfile that is now on main and develop branch

RemDelaporteMathurin, 2020-11-25:
I tried yesterday and the same error occured

RemDelaporteMathurin, 2021-02-10:
I can now build the image with windows. Closing this as it must have been solved along the way

closed 2021-02-10 by RemDelaporteMathurin