ukaea/paramak #515: making a better divertor for ball reactor

Shimwell, 2020-11-17:
Currently we have a diverter envelop that connects to the blanket

It works reasonably for divertor_radial_thickness which result in it being above the plasma but for lower divertor_radial_thickness it has a mismatch with the blanket connection

Here is an example of 26, 50 and 100 for the divertor radial thickness

```python
import paramak
my_reactor = paramak.BallReactor(
        inner_bore_radial_thickness=10,
        inboard_tf_leg_radial_thickness=30,
        center_column_shield_radial_thickness=60,
        divertor_radial_thickness=100,
        inner_plasma_gap_radial_thickness=30,
        plasma_radial_thickness=300,
        outer_plasma_gap_radial_thickness=30,
        firstwall_radial_thickness=30,
        blanket_radial_thickness=50,
        blanket_rear_wall_radial_thickness=30,
        elongation=2,
        triangularity=0.55,
        number_of_tf_coils=16,
        rotation_angle=180,
        pf_coil_radial_thicknesses=[50, 50, 50, 50],
        pf_coil_vertical_thicknesses=[50, 50, 50, 50],
        pf_coil_to_rear_blanket_radial_gap=50,
        pf_coil_to_tf_coil_radial_gap=50,
        outboard_tf_coil_radial_thickness=100,
        outboard_tf_coil_poloidal_thickness=50
)
my_reactor.export_stp()
```

```python
import paramak
my_reactor = paramak.BallReactor(
        inner_bore_radial_thickness=10,
        inboard_tf_leg_radial_thickness=30,
        center_column_shield_radial_thickness=60,
        divertor_radial_thickness=50,
        inner_plasma_gap_radial_thickness=30,
        plasma_radial_thickness=300,
        outer_plasma_gap_radial_thickness=30,
        firstwall_radial_thickness=30,
        blanket_radial_thickness=50,
        blanket_rear_wall_radial_thickness=30,
        elongation=2,
        triangularity=0.55,
        number_of_tf_coils=16,
        rotation_angle=180,
        pf_coil_radial_thicknesses=[50, 50, 50, 50],
        pf_coil_vertical_thicknesses=[50, 50, 50, 50],
        pf_coil_to_rear_blanket_radial_gap=50,
        pf_coil_to_tf_coil_radial_gap=50,
        outboard_tf_coil_radial_thickness=100,
        outboard_tf_coil_poloidal_thickness=50
)
my_reactor.export_stp()
```

```python
import paramak
my_reactor = paramak.BallReactor(
        inner_bore_radial_thickness=10,
        inboard_tf_leg_radial_thickness=30,
        center_column_shield_radial_thickness=60,
        divertor_radial_thickness=26,
        inner_plasma_gap_radial_thickness=30,
        plasma_radial_thickness=300,
        outer_plasma_gap_radial_thickness=30,
        firstwall_radial_thickness=30,
        blanket_radial_thickness=50,
        blanket_rear_wall_radial_thickness=30,
        elongation=2,
        triangularity=0.55,
        number_of_tf_coils=16,
        rotation_angle=180,
        pf_coil_radial_thicknesses=[50, 50, 50, 50],
        pf_coil_vertical_thicknesses=[50, 50, 50, 50],
        pf_coil_to_rear_blanket_radial_gap=50,
        pf_coil_to_tf_coil_radial_gap=50,
        outboard_tf_coil_radial_thickness=100,
        outboard_tf_coil_poloidal_thickness=50
)
my_reactor.export_stp()
```

[image]

[image]

[image]

Shimwell, 2020-11-17:
I think we need a minimal example of this intersection between a cylinder shape and a spline shape to show the resulting geometry is not where it should be. Then perhaps as CQ devs

RemDelaporteMathurin, 2020-11-18:
Fixed in #516

PR #516 (2020-11-17): offset from plasma consistent accross components
## Proposed changes

This small PR fixes #515 by making the offset from plasma of blanket, first wall and rear wall in BallReactor consistent with that of the divertor.

## Types of changes

What types of changes does your code introduce to the Paramak?
_Put an `x` in the boxes that apply_

- [x] Bugfix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Code refactoring
- [ ] Docume [... cut]
closed 2020-11-18 by RemDelaporteMathurin