ukaea/paramak #521: additional argument for port cutters

Shimwell, 2020-11-18:
The PortCutterRectangular and PortCutterCircular are great and make nice shapes.

One optional upgrade for us to consider is allowing them to have a inner raduis where the shape starts.

This would be useful when they are used to make a port in the outboard firstwall. One schenarios is users make a firstwall and then want to cut a port. Unfortunately the firstwall will be cut on the inboard and outboard. This is because the PortCutters start from 0 on the x axis.

Another argument for the inner radius could solve this use case problem and make the components even more useful.

RemDelaporteMathurin, 2020-11-18:
To be clear, you'd like to reproduce these extrusions on the VV ?
[image]

Is so, I think more than additional arguments, we need new shapes.

Shimwell, 2020-11-18:
I was planning on doing a large rectangle port then cut away a small rectangle.

To begin with I was just going to use it on relatively flat surfaces so I can probably get away with just a modified port cutter shape that accepts an inner_radius argument (defaults to 0)

In the longer term the curve of the vessel would be good to use as a cutter. The outer solid can be obtained using a cadquery selector (nearest point or perhaps largest volume)

Shimwell, 2020-11-19:
[image]

P-S-S-M, 2020-11-19:
Port within the paramak such as in the fantastic drawing above would be very useful. I think it will allow people to visualise the loss of breeding volume due to equatorial ports and look at their impact in simplified analysis.

Sam

billingsley-john, 2020-11-20:
Perhaps a combination of a cut and an intersect would achieve the desired port wholly within the correct outboard component

closed 2020-11-24 by RemDelaporteMathurin