ukaea/paramak #533: Submersion tokamak divertor construction error

billingsley-john, 2020-11-19:
Known error:
elongation=2.01
triangularity=0.51
Divertor is not constructed correctly. Cylindrical envelope used to define the divertor does not correctly intersect with the inboard and outboard first wall. Perhaps the inboard and outboard firstwall are not unioning correctly.

Alternative fix is to use the full coverage parametric component instead of separate inboard and outboard

billingsley-john, 2020-11-19:
[image]

Shimwell, 2020-12-16:
Is this one solved now, I lost track but thought it was working for certain dimentions