ukaea/paramak #545: Add option to build Dockerfile without the paramak source code as an option.

Shimwell, 2020-11-20:
The Dockerfile builds with the paramak source code everytime

We need an environment / dependency image for the circle ci testing and it would be good to keep this image upto date

We can keep it upto date with a new github action

The Dockerfile will need another optional build arg that allos the COPY of paramak source code, examples

Shimwell, 2020-11-22:
Looks like this is trickier to do that I though. The COPY command can't be put into and if statement like we have done elsewhere

Shimwell, 2020-12-01:
I can make another dockerfile in the paramak repo. 

- The filename dockerfile_dependencies and it will be quite similar in contents to the existing dockerfile. 
- It will differ from the existing dockerfile by not having the COPY commands
- It will also have an associated github actions .yml file that rebuilds it whenever we merge to main and publishes it with a ukaea/paramak_dependencies tag

RemDelaporteMathurin, 2020-12-01:
how about trying the optional copy command first ?

Shimwell, 2021-01-01:
Tried the optional copy command but it didn't work so made a new dockerfile and a PR #647

closed 2021-01-01 by Shimwell