ukaea/paramak #548: Error with circular ports rotated 180°

RemDelaporteMathurin, 2020-11-20:
How to reproduce:

```
import paramak

test_component = paramak.PortCutterCircular(
    distance=1,
    z_pos=0,
    radius=1,
    azimuth_placement_angle=[0, 180],
    #azimuth_placement_angle=[90, 270],
)

test_component.export_stl('out.stl')
```

RemDelaporteMathurin, 2020-11-21:
Moreover, I've isolated the bug to pure cadquery commands and it seems it only occurs when workplane is "XZ".
Perhaps a good issue for CQ ?
```python
import cadquery as cq

def create_solid(workplane):
    solid0 = (
        cq.Workplane(workplane, 
        .moveTo(0, 0)
        .circle(1)
        .extrude(1, both=False)
    )

    azimuth_placement_angles = [0, 180]

    solid1 = solid0.rotate((0, 0, -1), (0, 0, 1), 0)
    solid2 = solid0.rotate((0, 0, -1), (0, 0, 1), 180)
    solid = cq.Workplane(workplane)

    solid = solid.union(solid1)
    solid = solid.union(solid2)
    return solid

solid = create_solid("XY") # works
solid = create_solid("XZ") # does not work
solid = create_solid("YZ") # does not work
```

RemDelaporteMathurin, 2020-11-21:
Funny enough, the following seems to work

```python
import cadquery as cq

def create_solid(workplane):
    solid0 = (
        cq.Workplane(workplane)
        .moveTo(0, 0)
        .circle(1)
        .extrude(1, both=False)
    )

    solid1 = solid0.rotate((0, 0, -1), (0, 0, 1), 0)
    solid2 = solid0.rotate((0, 0, -1), (0, 0, 1), 180)
    solid = cq.Workplane(workplane)

    solid = solid.union(solid1)
    solid = solid.union(solid2, clean=False)
    return solid

solid = create_solid("XY") # works
solid = create_solid("XZ") # works
solid = create_solid("YZ") # works
```

billingsley-john, 2020-11-23:
This works locally on CadQuery 2.0, so may be an issue with CadQuery master

RemDelaporteMathurin, 2020-11-23:
I opened an issue in CQ https://github.com/CadQuery/cadquery/issues/519

billingsley-john, 2020-11-23:
Note: This issue is not isolated to the port cutters. The issue also occurs in the underlying ExtrudeCircleShape class

closed 2020-11-23 by RemDelaporteMathurin