ukaea/paramak #551: CenterColumnStudy reactor broken

RemDelaporteMathurin, 2020-11-21:
The following code produces a non-matching divertor:

```python
import paramak

test_reactor = paramak.CenterColumnStudyReactor(
    inner_bore_radial_thickness=20,
    inboard_tf_leg_radial_thickness=50,
    center_column_shield_radial_thickness_mid=50,
    center_column_shield_radial_thickness_upper=100,
    inboard_firstwall_radial_thickness=20,
    divertor_radial_thickness=100,
    inner_plasma_gap_radial_thickness=80,
    plasma_radial_thickness=200,
    outer_plasma_gap_radial_thickness=90,
    elongation=2.3,
    triangularity=0.45,
    plasma_gap_vertical_thickness=40,
    center_column_arc_vertical_thickness=520,
    rotation_angle=180)

test_reactor.export_stl("out")
```

[image]

RemDelaporteMathurin, 2020-11-21:
I think this was the case since the beginning of this shape but is more significant with big divertors.

Shall I close this since this Reactor is focused on the Center column ?
[image]

Shimwell, 2020-11-24:
I think that I understand the problem. We have a divertor that is just an envelope and on the ball reactor and the submersion reactor this connects to a flat center column. This divertor in this reactor connects to a curved center column. The connection is harder to get right as there are more connection scenarios.

- The center column and divertor connect entirely on a vertical flat surface (easy option same as ball and sub reactor)

- The center column and divertor connect on the part of the center column that is curved

- The center column and divertor connect on the part of the center column that is flat and curved 

So perhaps the solution is to use a different divertor cutting shape. It currently gets cut with a basic cylinder. However we could cut it with a union of the first wall and the center column shield

RemDelaporteMathurin, 2020-11-24:
> So perhaps the solution is to use a different divertor cutting shape. It currently gets cut with a basic cylinder. However we could cut it with a union of the first wall and the center column shield

currently the divertor is an intersect of blanket and cylinder.

I think the problem here isn't the cylinder, but rather the blanket that doesn't connect to the flat area of the inboard first wall. 

Intersect the blanket with the center parts (a union of the first wall and the center column shield) will produce a weird thing at the center
[image]