ukaea/paramak #571: Class diagram needed for docs / paper

Shimwell, 2020-11-24:
A while ago I am sure we made a couple of class diagrams. 

One was on mathch.com and that was not very good and there was another one somewhere that @RemDelaporteMathurin made. 

Would be good to track these down perhaps update them and add the diagram to the docs / paper.

RemDelaporteMathurin, 2020-11-24:
Are you talking about the diagram of interactions of the paramak with other codes ?

Shimwell, 2020-11-24:
Not the interaction with other codes one

I thought we made a diagram of the class structure within the paramak

RemDelaporteMathurin, 2020-11-25:
Ah that rings a bell.

RemDelaporteMathurin, 2020-11-25:
https://lucid.app/invitations/accept/7a624a34-7695-4ca4-9544-c3c8887c0eaf

This one

openmcworkshop, 2020-11-25:
Yes that is the one, thanks for finding it 

[image]

openmcworkshop, 2020-11-25:
I'm keen to use something like this html image map to make a clickable map in the docs
https://www.w3schools.com/html/html_images_imagemap.asp

Shimwell, 2020-11-25:
Yes that is the one, thanks for finding it
[image]

Shimwell, 2020-11-25:
I'm keen to use something like this html image map to make a clickable map in the docs
https://www.w3schools.com/html/html_images_imagemap.asp

Shimwell, 2020-12-01:
Feedback from the paper review
"This might be a bit much, but would a class hierarchy diagram benefit this section? It would quickly show how the higher level Parametric and Reactor Class components inherit from the base Shape class and build upon each other."