ukaea/paramak #578: Inconsistency in argument names across BlanketCutter shapes

RemDelaporteMathurin, 2020-11-25:
`BlanketCutterParallel `has a `thickness ` argument, whereas `BlanketCutterStar `has a `distance `argument. 

What should we do ? @Shimwell @billingsley-john

Shimwell, 2020-11-25:
I vote distance in both cases as this is the argument name in the extrude shapes.

RemDelaporteMathurin, 2020-11-25:
True, but the user could be confused as there are many distances:
[image]

Distance is also suitable for width or height. In ExtrudeShape, there is not doubt cause there is only one extrusion distance possible.

billingsley-john, 2020-11-26:
I think that whatever word is used, whether that be 'width', 'length', 'thickness', 'distance', there will be some ambiguity. Like @Shimwell, I like the use of 'distance' for consistency with the extrude shapes as that parameter is the distance extruded by the function

RemDelaporteMathurin, 2020-11-26:
Ok then. The only potential issue I find is that the `distance `parameter from the Cutter class and from the inherited class don't have the same value. 

This may mess with the setters /getters being overriden