ukaea/paramak #587: Breeding module models would be very useful for T transport analysis

RemDelaporteMathurin, 2020-11-26:
T transport simulations in BBs require CAD files for creating a mesh.
It'd be really useful to have these in the Paramak including First Wall, Breeding zones, cooling channels, structural parts (EUROFER).

Pretty much as what is available here: https://github.com/ukaea/breeder_blanket_model_maker

Shimwell, 2020-11-26:
This can be broken down into a few stages.

First the segmentedBallReactor needs improving, part of the functionality needed is currently in the example script and should be moved into the segmentedBallReactor class
https://github.com/ukaea/paramak/blob/f807d88098458d1975dd1a4a47dcea22d9f74785/examples/example_parametric_reactors/segmented_blanket_ball_reactor.py#L40-L52

Secondly the segment_gap between BlanketFPPoloidalSegments needs to change a bit so that top and bottom caps can be made using the method below. Currently just the top caps are made

```python
import paramak
import numpy as np

cutter_small = paramak.BlanketCutterStar(
    distance=50)

cutter_big = paramak.BlanketCutterStar(
    distance=100)

componentsmall = paramak.BlanketFPPoloidalSegments(
    thickness=80,
    start_angle=90,
    stop_angle=-90,
    num_segments=5,
    rotation_angle=180,
    offset_from_plasma=40,
    cut=cutter_big,
    segments_gap=40
)

componentbig = paramak.BlanketFPPoloidalSegments(
    thickness=100,
    start_angle=90,
    stop_angle=-90,
    num_segments=5,
    rotation_angle=180,
    offset_from_plasma=20,
    segments_gap=20,
    cut=[cutter_small, componentsmall]
)

componentbig.export_stp('casing.stp')
```
[image]

Shimwell, 2020-11-26:
There are additional stages after this depending on the design of blanket but the filleted wrap around firstwall with top and bottom caps is a good start.

Shimwell, 2021-02-08:
Small amount of progress on this fron as we have a PR for hex shaped pins open #706